qmk/qmk_toolbox #224: Crash when flashing on Mojave (10.14)

pgooch, 2020-09-16:
## Describe the Bug

While in the past I was able to flash things without trouble now whenever it starts flashing (either by connecting with auto-flash or clicking "Flash") the program quits. I am no longer seeing the system error pop up when this happens.

## System Information

 - Operating system: 10.14.6
 - QMK Toolbox version: 0.0.19

## Additional Context

I went diving into the console app and found some records from the last failure, they don't seem to be particularly helperfull but I'll add them below. I previously flashed a 4 switch macro pad and while it didn't crash and the flashing seemed to work the device didn't work and plugging it in caused other issues with USB on the machine (my DAC would pop in and out for a while before the macropad eventually gave up). I figured it was just something with the pad at the time but now I'm not so sure.

```
error	16:18:37.359196 -0700	QMK Toolbox	        AEGetDescData of non-desc type 'reco' not a good idea
default	16:18:37.417907 -0700	QMK Toolbox	Faulting in NSHTTPCookieStorage singleton
default	16:18:37.417924 -0700	QMK Toolbox	Faulting in CFHTTPCookieStorage singleton
default	16:18:37.417934 -0700	QMK Toolbox	Creating default cookie storage with process/bundle identifier
default	16:18:37.419110 -0700	QMK Toolbox	TIC TCP Conn Start [1:0x600000abb000]
default	16:18:37.419303 -0700	QMK Toolbox	[C1 Hostname#571fa8e7:80 tcp, url hash: 8479fe79] start
default	16:18:37.419956 -0700	QMK Toolbox	nw_connection_report_state_with_handler_locked [C1] reporting state preparing
default	16:18:37.420595 -0700	QMK Toolbox	Task <B0F821F7-3F05-475D-820A-26D0A137F181>.<0> setting up Connection 1
default	16:18:37.523884 -0700	QMK Toolbox	nw_socket_handle_socket_event [C1.1:2] Socket received CONNECTED event
default	16:18:37.524000 -0700	QMK Toolbox	nw_endpoint_flow_protocol_connected [C1.1 IPv4#940be249:80 in_progress socket-flow (satisfied)] Transport protocol connected
default	16:18:37.524121 -0700	QMK Toolbox	nw_endpoint_flow_protocol_connected [C1.1 IPv4#940be249:80 in_progress socket-flow (satisfied)] Output protocol connected
default	16:18:37.524356 -0700	QMK Toolbox	nw_connection_report_state_with_handler_locked [C1] reporting state ready
default	16:18:37.524728 -0700	QMK Toolbox	TIC TCP Conn Connected [1:0x600000abb000]: Err(16)
default	16:18:37.524859 -0700	QMK Toolbox	TIC TCP Conn Event [1:0x600000abb000]: 1
default	16:18:37.525102 -0700	QMK Toolbox	Task <B0F821F7-3F05-475D-820A-26D0A137F181>.<0> now using Connection 1
default	16:18:37.526378 -0700	QMK Toolbox	Task <B0F821F7-3F05-475D-820A-26D0A137F181>.<0> sent request, body N
default	16:18:37.625108 -0700	QMK Toolbox	Task <B0F821F7-3F05-475D-820A-26D0A137F181>.<0> received response, status 200 content K
default	16:18:37.724104 -0700	QMK Toolbox	Task <B0F821F7-3F05-475D-820A-26D0A137F181>.<0> response ended
default	16:18:37.724212 -0700	QMK Toolbox	Task <B0F821F7-3F05-475D-820A-26D0A137F181>.<0> done using Connection 1
default	16:18:38.765996 -0700	QMK Toolbox	AggregateDictionary is not supported on this platform
error	16:18:39.424041 -0700	QMK Toolbox	An uncaught exception was raised
error	16:18:39.424071 -0700	QMK Toolbox	<private>
error	16:18:39.424117 -0700	QMK Toolbox	<private>
```
The keyboard I'm trying to flash is a  wheatfield split 75 clone purchased off Aliexpress. I initially flashed it with the toolbox and it worked fine. That was on my last OS installation and I've since moved back to Mojave. I have not gotten a successful flash on it since I did that (although I doubt that's directly related).

fauxpark, 2020-09-17:
Looks like bootloadHID didn't have the execute bit set...

If you run `chmod +x /Applications/QMK\ Toolbox.app/Contents/Resources/bootloadHID` that should fix it.

closed 2020-09-17 by fauxpark

pgooch, 2020-09-17:
Thanks, that sorted it.

PR #225 (2020-09-17): [macOS] Fix file permissions for bootloadHID

## Description

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #224