qmk/qmk_toolbox #227: macOS cannot check QMK Toolbox for malware, cannot open

GalacticLion7, 2020-09-18:
On macOS, QMK Toolbox cannot open:

[image]

* macOS Version: 10.15.6
* QMK Toolbox Version: 0.0.20

And yes, I know that I could just override it in System Preferences, but could you not fix it from your end so it works for everyone?

skullydazed, 2020-09-18:
Signing the app is something we keep kicking down the road. We will do it, but I can't give you a timeline on it.

In the meantime you can bypass this check without changing your global preferences by right clicking QMK Toolbox and clicking "Open". After that you will be able to open the app in the normal fashion.

focusaurus, 2020-10-26:
Note on macos 10.15.7 I cannot open this even with the finder right click open workaround. I believe this is stock macos behavior now although there's some chance it's extra lockdown by my workplace's IT setup.

kyusu, 2020-11-24:
> Note on macos 10.15.7 I cannot open this even with the finder right click open workaround. I believe this is stock macos behavior now although there's some chance it's extra lockdown by my workplace's IT setup.

It's not your setup. I can't open it either on my (private) Mac.

skullydazed, 2020-11-24:
For the time being you can [work around this](https://support.apple.com/en-us/HT202491) by opening System Preferences, going to Security & Privacy, and under the General tab you should see QMK Toolbox listed. Click "Open Anyway" to allow QMK Toolbox to open now and in the future.

[image]

kyusu, 2020-11-24:
Interestingly enough only using this workaround did not work for me. What did work was doing the above and cleaning the quarantine attribute with `sudo xattr -cr /Applications/QMK\ Toolbox.app` ([https://ss64.com/osx/xattr.html](xattr)).

focusaurus, 2020-11-25:
I don't get that "Open Anyway" button in the security & privacy screen at all. shrug. but @kyusu that `xattr` command line did work so now I can launch the toolbox. Thanks!

kbrock, 2021-03-25:
Running High Sierra and Big sur. The following works for me.

Double click the application. It says "no go" but offers to show me the application in the finder.
It opens the Applications folder and highlights the application.
I right click the application and choose open (yes, seems the same as just double clicking it)
It now gives me 3 options, one of which is open anyway.

I do not need to go into the security and privacy settings.
This is a personal laptop, but it has worked in reasonably locked down corporate laptops as well.

HTH

amowry, 2021-05-21:
Just another request for signing the QMK Toolbox :) app. I love it and recommend it to my customers, but the fact that many of them have trouble installing it does cause me a lot of headaches :(. Thanks!

rfong, 2021-06-06:
> Interestingly enough only using this workaround did not work for me. What did work was doing the above and cleaning the quarantine attribute with `sudo xattr -cr /Applications/QMK\ Toolbox.app` ([https://ss64.com/osx/xattr.html](xattr)).

`-c` recursively removes *all* `xattr` attributes. It's usually not a concern, but to be safe I'd recommend just specifying the attr causing the annoyance. `xattr -d com.apple.quarantine /Applications/QMK\ Toolbox.app`

daprice, 2021-12-07:
Here’s yet another request to sign the app to address this. Considering that most people probably type passwords, account numbers, and other private info using keyboards flashed using QMK Toolbox, it seems important to provide the ability to confirm that one’s copy of the app hasn’t been infected with malware during or after download.

Users of other similarly-sized open source projects that didn’t sign their app [have already been targeted by attacks that might have been mitigated if the app was signed](https://blog.malwarebytes.com/threat-analysis/mac-threat-analysis/2017/05/handbrake-hacked-to-drop-new-variant-of-proton-malware/).

nooges, 2022-05-18:
I've started doing some preliminary work on having the app signed and notarized now through GitHub Actions, just a bunch of tedious work right now adding steps to the workflow, running it, and waiting for the result.

amowry, 2022-05-19:
Great, thanks so much for your work on this!

pkrakow, 2023-03-11:
I was able to get this to work on MacOS Ventura 13.1 earlier today. The series of steps was slightly different from what's discussed above (I had to click through a second dialog warning) but the app ultimately worked.

stevethecollier, 2023-03-14:
another approach is to "show in finder", right click, select open from the file options, and then you will be given the option to start the program.

Kl-11, 2023-10-25:
Any chance this has been resolved in the beta or planned to? Now Nuphy shifted to VIA/QMK and more mainstream users like myself are figuring this out. It would be much appreciated if you can sign the app and provide a simpler / safer route for us to use this. Great work overall, new to the mechanical keyboard world.

nooges, 2023-10-25:
Due to the deletion of the GitHub Action from the marketplace that we were using to sign Mac build with, we had to remove signing for now (https://github.com/qmk/qmk_toolbox/pull/420). Not sure yet when I'll get around to fixing the signing issue by finding new actions from the marketplace to work with.

Kl-11, 2023-10-25:
Thanks nooges. Hopefully an alternative comes along soon. Appreciate your efforts. Cheers

tzarc, 2026-08-19:
Fixed by https://github.com/qmk/qmk_toolbox/pull/511

closed 2026-08-19 by tzarc