qmk/qmk_toolbox #230: Crashing on MacOS 10.15.7

franklinharvey, 2020-09-26:
Whenever I click "flash" the app crashes

## Describe the Bug

## System Information

 - Operating system: MacOS 10.15.7
 - QMK Toolbox version: Latest: 0.0.20

## Additional Context
here's the crash report
```
Process:               QMK Toolbox [19594]
Path:                  /Applications/QMK Toolbox.app/Contents/MacOS/QMK Toolbox
Identifier:            fm.qmk.toolbox
Version:               0.0.20 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QMK Toolbox [19594]
User ID:               501

Date/Time:             2020-09-26 17:57:08.108 -0600
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Anonymous UUID:        FA0213FB-A3BD-295B-1A33-1E8EADAD7049

Sleep/Wake UUID:       DE69A02C-7156-4550-9128-FD310D577E79

Time Awake Since Boot: 99000 seconds
Time Since Wake:       1900 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderArray initWithObjects:count:]: attempt to insert nil object from objects[7]'
abort() called
terminating with uncaught exception of type NSException
```

fauxpark, 2020-09-27:
That log unfortunately doesn't give me much to go on. Could you try to reproduce the crash using this debug build?
[QMK Toolbox.app.zip](https://github.com/qmk/qmk_toolbox/files/5287781/QMK.Toolbox.app.zip)

franklinharvey, 2020-09-27:
Sure, here's when I filter by QMK Toolbox. I am trying to flash a Discipline65, which I had done in the past.

```
error	17:23:55.874511-0600	QMK Toolbox	An uncaught exception was raised
error	17:23:55.874577-0600	QMK Toolbox	*** -[__NSPlaceholderArray initWithObjects:count:]: attempt to insert nil object from objects[7]
error	17:23:55.875260-0600	QMK Toolbox	(
	0   CoreFoundation                      0x00007fff36f86727 __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x00007fff6fe8ea9e objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff37035796 _CFThrowFormattedException + 194
	3   CoreFoundation                      0x00007fff37033ab3 -[__NSPlaceholderArray initWithCapacity:].cold.1 + 0
	4   CoreFoundation                      0x00007fff36e8a44a -[__NSPlaceholderArray initWithObjects:count:] + 154
	5   CoreFoundation                      0x00007fff36ee6015 +[NSArray arrayWithObjects:count:] + 40
	6   QMK Toolbox                         0x000000010974b133 QMK Toolbox + 33075
	7   QMK Toolbox                         0x0000000109749cb0 QMK Toolbox + 27824
	8   QMK Toolbox                         0x000000010974be5f QMK Toolbox + 36447
	9   libdispatch.dylib                   0x00007fff70fd5658 _dispatch_client_callout + 8
	10  libdispatch.dylib                   0x00007fff70fd7818 _dispatch_continuation_pop + 414
	11  libdispatch.dylib                   0x00007fff70fe74be _dispatch_source_invoke + 2084
	12  libdispatch.dylib                   0x00007fff70fe0b6d _dispatch_main_queue_callback_4CF + 618
	13  CoreFoundation                      0x00007fff36f49b23 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	14  CoreFoundation                      0x00007fff36f097aa __CFRunLoopRun + 2042
	15  CoreFoundation                      0x00007fff36f08953 CFRunLoopRunSpecific + 466
	16  HIToolbox                           0x00007fff35b23abd RunCurrentEventLoopInMode + 292
	17  HIToolbox                           0x00007fff35b237d5 ReceiveNextEventCommon + 584
	18  HIToolbox                           0x00007fff35b23579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
	19  AppKit                              0x00007fff34169039 _DPSNextEvent + 883
	20  AppKit                              0x00007fff34167880 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
	21  AppKit                              0x00007fff3415958e -[NSApplication run] + 658
	22  AppKit                              0x00007fff3412b396 NSApplicationMain + 777
	23  libdyld.dylib                       0x00007fff7102ecc9 start + 1
)
```

franklinharvey, 2020-09-28:
sorry this is my bad. I still had my Pocket Programmer plugged in while trying this. It works perfectly fine when thats unplugged

closed 2020-09-28 by franklinharvey

fauxpark, 2020-09-28:
@franklinharvey I'd still be interested in seeing that .crash file, the Toolbox should not be outright dying when some other device is plugged in.

fauxpark, 2020-09-28:
For reference, I've symbolicated the three QMK Toolbox lines in the log above:
```
	6   QMK Toolbox                         0x000000010974b133 -[Flashing flashUSBTiny:withFile:] (in QMK Toolbox) (Flashing.m:178)
	7   QMK Toolbox                         0x0000000109749cb0 -[Flashing flash:withFile:] (in QMK Toolbox) (Flashing.m:70)
	8   QMK Toolbox                         0x000000010974be5f __32-[AppDelegate flashButtonClick:]_block_invoke (in QMK Toolbox) (AppDelegate.m:68)
```
I don't see anything particularly egregious about `flashUSBTiny:withFile:` though.