qmk/qmk_toolbox #233: QMK Toolbox for Web using WebHID?

bbeckford, 2020-10-07:
I don't know a lot about how QMK works, but I thought it might be worthwhile sharing the news on WebHID in case it's helpful or interesting here. Sorry if it's been posted already!

Could WebHID enable a web-based version of QMK toolbox, maybe built-in to the QMK Configurator?

Useful links:

- https://youtu.be/GNuG-5m4Ud0?list=WL&t=102
- https://developers.google.com/web/updates/2020/10/nic86#web-hid
- https://developers.chrome.com/origintrials/#/view_trial/1074108511127863297
- https://github.com/WICG/webhid/blob/master/EXPLAINER.md

fauxpark, 2020-10-07:
I'm not too sure there's much of a point. Most of the supported bootloaders require direct USB access to flash, and in fact we need to run on the host because of the bundled binaries (dfu-programmer, dfu-util, avrdude, etc.). The only things we need access to the HID device for are console logging, and the BootloadHID bootloader (but we use the bootloadHID flasher binary anyway).

WebHID could perhaps be useful for some application of the Raw HID feature in QMK, but apart from that...dunno.

bbeckford, 2020-10-13:
Thanks for the response - shame it's not more useful right now!

closed 2020-10-13 by bbeckford