qmk/qmk_toolbox #234: When using --force memory fails to validate

TheRubberPig, 2020-10-09:
## Describe the Bug

The RGB headers on my board have recently been freaking out so I went to re flash the board and saw there was a memory error and to use --force. After updating to QMK 0.0.20 when it tries to force the flash it now fails to validate the memory on the board. Some googling suggest performing an ISP flash which I want to avoid if possible due to the complexity of it, is there a way to dump the memory within QMK before performing the flash?

## System Information

 - Operating system: Windows 10 v10.0.19041 Build 19041
 - QMK Toolbox version: 0.0.20

## Additional Context

The log info from the failed flash is below.

*** Attempting to flash, please don't remove device
>>> dfu-programmer.exe atmega32u4 erase --force
    Erasing flash...  Success
>>> dfu-programmer.exe atmega32u4 flash --force "C:\Users\Dave\Desktop\Macro Pad\dmqdesign_spin_TheRubberPig.hex"
    Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
    0%                            100%  Programming 0x6480 bytes...
>>> dfu-programmer.exe atmega32u4 reset
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    0%                            100%  Reading 0x7000 bytes...
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    Validating...  ERROR
    25630 invalid bytes in program region, 2932 outside region.
    FAIL
    Memory did not validate. Did you erase?
*** Atmel DFU device disconnected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)

DRWRammy, 2021-01-17:
I am currently having a very similar issue when trying to flash my keyboard, however I did not have any issues with RGB headers or notice anything else. I managed to successfully flash it couple of days ago, without any problems. I just wanted to restore default layout from QMK configurator for my DZ60rgb v2 ansi board and it failed. Tried flashing another .hex file from VIA firmware and it failed again. Currently I can configure my layout on VIA without any issue however I am not able to flash anything else on the board using QMK toolbox.

**System Info**
- OS Windows 10 Pro N OS build 19041.746 version 2004
- QMK Toolbox version: 0.0.20

>>> *** Atmel DFU device connected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)
*** Attempting to flash, please don't remove device
>>> dfu-programmer.exe atmega32u4 erase --force
    Erasing flash...  Success
    Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
>>> dfu-programmer.exe atmega32u4 flash --force "D:\Mano dokumentai\QMK keyboard layout files\dztech_dz60rgb_ansi_v2_via.hex"
    0%                            100%  Programming 0x6000 bytes...
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    0%                            100%  Reading 0x7000 bytes...
>>> dfu-programmer.exe atmega32u4 reset
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    Validating...  ERROR
    24467 invalid bytes in program region, 4080 outside region.
    FAIL
    Memory did not validate. Did you erase?
*** Atmel DFU device disconnected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)

DRWRammy, 2021-01-30:
Quick update: flashing ISP fixed my problem. I suspect that my memory got corrupted since I had previously encountered keyboard crash/hang up on Windows boot.

joevegan, 2021-02-15:
Same issue here. Also my if I currently press RBG mode + 5 times the board becomes unresponsive and I have to press the physical reset button on it to get it into DFU then cancel out if it, or attempt to flash but I get the same x invalid bytes, y outside region. Flashing or clearing EEPROM doesn't change the firmware on the board either.

it is also a DZ60 RGB v2 ANSI PCB
using version: QMK Toolbox version: 0.0.20

chuan, 2021-03-09:
I am hitting the same issue with DZ60rgb v2 ansi board using qmk toolbox 0.0.21.

NYUTerminal, 2021-03-17:
@chuan I am having the same issue on DZ60rgb v2 ansi board using qmk toolbox 0.0.21. The first time flashing the firmware was successful but after that I keep getting `Memory did not validate. Did you erase?` how did you do the ISP flashing?

chuan, 2021-03-17:
> @chuan I am having the same issue on DZ60rgb v2 ansi board using qmk toolbox 0.0.21. The first time flashing the firmware was successful but after that I keep getting `Memory did not validate. Did you erase?` how did you do the ISP flashing?

I used [SparkFun PocketAVR ](https://www.sparkfun.com/products/9825) and followed instructions here: https://beta.docs.qmk.fm/using-qmk/guides/keyboard-building/isp_flashing_guide

proov, 2021-04-22:
> I used [SparkFun PocketAVR ](https://www.sparkfun.com/products/9825) and followed instructions here: https://beta.docs.qmk.fm/using-qmk/guides/keyboard-building/isp_flashing_guide

I'm having this issue too, when i try to flash it, it seems working but on the last validation step it fails :
[image]

Im lucky that my board is still working fine, but i can't flash it anymore to customise some shortcuts.

Is there any software solution ? i would like to avoid the « ISP flashing », i don't even know what it is 😅

NYUTerminal, 2021-04-22:
I tried the ISP flashing with SparkFun PocketAVR but it didn't workout .. my board is still working but I cannot flash any new firmware at this point. I had to get a new board. I might set it up later.

CarbonMac, 2021-04-29:
One day into having this board and have to buy another part :/ lame. This is a dz60 issue for sure.

fauxpark, 2021-04-29:
Sounds like you ISP flashed the firmware and not the bootloader. If you only flash the firmware, there is no bootloader to allow you to flash over USB.