qmk/qmk_toolbox #243: option requires an argument -- P

valdiieee, 2020-12-15:
Hi im trying to flash firmware for my handwired keyboard but it always says option requires an argument -- P, hiw to fix it?
*** Caterina device connected (usbser): Arduino LLC (www.arduino.cc) Arduino Micro bootloader (COM4) (2341:0037:0001)
*** Attempting to flash, please don't remove device
>>> avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"C:\Users\JAZZ FOR YOUR SOUL\qmk_firmware\.build\makrosu_default.hex":i -P 
    C:\Users\JAZZ FOR YOUR SOUL\AppData\Local\QMK\QMK Toolbox\0.0.20\avrdude.exe: option requires an argument -- P
    Usage: avrdude.exe [options]
    Options:
      -p <partno>                Required. Specify AVR device.
      -b <baudrate>              Override RS-232 baud rate.
      -B <bitclock>              Specify JTAG/STK500v2 bit clock period (us).
      -C <config-file>           Specify location of configuration file.
      -c <programmer>            Specify programmer type.
      -D                         Disable auto erase for flash memory
      -i <delay>                 ISP Clock Delay [in microseconds]
      -P <port>                  Specify connection port.
      -F                         Override invalid signature check.
      -e                         Perform a chip erase.
      -O                         Perform RC oscillator calibration (see AVR053). 
      -U <memtype>:r|w|v:<filename>[:format]
                                 Memory operation specification.
                                 Multiple -U options are allowed, each request
                                 is performed in the order specified.
      -n                         Do not write anything to the device.
      -V                         Do not verify.
      -u                         Disable safemode, default when running from a script.
      -s                         Silent safemode operation, will not ask you if
                                 fuses should be changed back.
      -t                         Enter terminal mode.
      -E <exitspec>[,<exitspec>] List programmer exit specifications.
      -x <extended_param>        Pass <extended_param> to programmer.
      -y                         Count # erase cycles in EEPROM.
      -Y <number>                Initialize erase cycle # in EEPROM.
      -v                         Verbose output. -v -v for more.
      -q                         Quell progress output. -q -q for less.
      -l logfile                 Use logfile rather than stderr for diagnostics.
      -?                         Display this usage.
    
    avrdude version 6.3, URL: <http://savannah.nongnu.org/projects/avrdude/>

fauxpark, 2020-12-15:
Make sure you're on the latest version: https://github.com/qmk/qmk_toolbox/releases

valdiieee, 2020-12-15:
> Make sure you're on the latest version: https://github.com/qmk/qmk_toolbox/releases

im using the latest version 0.0.20

valdiieee, 2020-12-15:
ah there are some mistakes in my rules.mk now it works.

closed 2020-12-15 by valdiieee