qmk/qmk_toolbox #247: Toolbox cannot open and freezes

Xelus22, 2021-01-11:
## Describe the Bug

If `CONSOLE_ENABLE` and `COMMAND_ENABLE` are both enabled, Toolbox will *NOT* open. It is shown as a background process in task manager and does show up in alt+tab (see image below).

If the keyboard has both of these enabled, and the keyboard is unplugged, Toolbox WILL open normally and display properly. However once the keyboard is re-plugged in after this, Toolbox will freeze and be unresponsive with the "not responding" option in task manager.

## System Information

 - Operating system: Windows 10
 - QMK Toolbox version: 0.0.19 and 0.0.20

## Additional Context

[image]

fauxpark, 2021-01-11:
I'm not able to reproduce this on Windows 10, with my Satan. Both 0.0.20 and current master.

Xelus22, 2021-01-11:
I have just tested this more thoroughly. When you close QMK toolbox with the "X" button in the top right corner, the next time you open toolbox it will freeze/not pop up. However if I go to task manager and "end task", the next time you open it will be perfectly normal.

fauxpark, 2021-01-11:
Would you be able to debug it in Visual Studio?

fauxpark, 2021-07-16:
@Xelus22 any update on this?

Xelus22, 2021-07-16:
I am no longer able to re-produce this issue. Not sure what was the cause

Xelus22, 2021-07-16:
Ill be closing this then.

nooges, 2021-08-18:
I just ran into this issue today, if I had my Iris plugged in (not sure which features are enabled in rules.mk), QMK Toolbox would stay in the background.

If the board was left unplugged and then QMK Toolbox is launched, it works fine. Didn't notice any freeze ups if the board was plugged in with QMK Toolbox fully launched.

Tested with 0.0.21, 0.1.0, and 0.1.1.

fauxpark, 2021-08-18:
@nooges if Console is enabled on the board, can you check whether #297 helps any?

nooges, 2021-08-18:
A bit more debugging, having the Satisfaction75 plugged in at the same time as the Iris does cause the issue. About to test with #297 in a bit.

nooges, 2021-08-18:
@fauxpark Tested with #297, looks like it works and both boards are detected.

fauxpark, 2021-08-19:
Good to hear. I suspect this problem has something to do with the HidLibrary `ReadReport()` callback and how it works/is used. My PR does a much better job of managing `HidDevice`s and their report callbacks, IMO.

Ardhimas, 2022-01-01:
Is there a workaround for this issue besides unplugging the keyboard before running QMK toolbox every time?

fauxpark, 2022-01-02:
It's already fixed as of the PR mentioned above. There just hasn't been a new release yet, there's some more improvements I haven't had time to finish.

closed 2022-01-02 by fauxpark