qmk/qmk_toolbox #248: installed qmk crashes on start up

timovp, 2021-01-16:
## Describe the Bug

I used the installer to install qmk toolbox and launched it. one day to the next it crashes on start up.
## System Information

 - Operating system: windows 10 pro
 - QMK Toolbox version: 0.0.20 installer version

## Additional Context
it happened over night, or let say over a power down power up cycle. 
event log displays these errors: 
```Toepassing: qmk_toolbox.exe
Framework-versie: v4.0.30319
Beschrijving: het proces is beëindigd als gevolg van een onverwerkte uitzondering.
Uitzonderingsinformatie: System.Xml.XmlException
   bij System.Xml.XmlTextReaderImpl.Throw(System.Exception)
   bij System.Xml.XmlTextReaderImpl.Throw(System.String, System.String[])
   bij System.Xml.XmlTextReaderImpl.ParseText(Int32 ByRef, Int32 ByRef, Int32 ByRef)
   bij System.Xml.XmlTextReaderImpl.ParseText()
   bij System.Xml.XmlTextReaderImpl.ParseElementContent()
   bij System.Xml.XmlTextReaderImpl.Read()
   bij System.Xml.XmlTextReader.Read()
   bij System.Configuration.XmlUtil.StrictReadToNextElement(System.Configuration.ExceptionAction)
   bij System.Configuration.BaseConfigurationRecord.ScanSectionsRecursive(System.Configuration.XmlUtil, System.String, Boolean, System.String, System.Configuration.OverrideModeSetting, Boolean)
   bij System.Configuration.BaseConfigurationRecord.ScanSectionsRecursive(System.Configuration.XmlUtil, System.String, Boolean, System.String, System.Configuration.OverrideModeSetting, Boolean)
   bij System.Configuration.BaseConfigurationRecord.ScanSections(System.Configuration.XmlUtil)
   bij System.Configuration.BaseConfigurationRecord.InitConfigFromFile()

Uitzonderingsinformatie: System.Configuration.ConfigurationErrorsException
   bij System.Configuration.ConfigurationSchemaErrors.ThrowIfErrors(Boolean)
   bij System.Configuration.BaseConfigurationRecord.ThrowIfParseErrors(System.Configuration.ConfigurationSchemaErrors)
   bij System.Configuration.BaseConfigurationRecord.ThrowIfInitErrors()
   bij System.Configuration.ClientConfigurationSystem.OnConfigRemoved(System.Object, System.Configuration.Internal.InternalConfigEventArgs)

Uitzonderingsinformatie: System.Configuration.ConfigurationErrorsException
   bij System.Configuration.ClientConfigurationSystem.OnConfigRemoved(System.Object, System.Configuration.Internal.InternalConfigEventArgs)
   bij System.Configuration.Internal.InternalConfigRoot.OnConfigRemoved(System.Configuration.Internal.InternalConfigEventArgs)
   bij System.Configuration.Internal.InternalConfigRoot.RemoveConfigImpl(System.String, System.Configuration.BaseConfigurationRecord)
   bij System.Configuration.BaseConfigurationRecord.GetSectionRecursive(System.String, Boolean, Boolean, Boolean, Boolean, System.Object ByRef, System.Object ByRef)
   bij System.Configuration.BaseConfigurationRecord.GetSection(System.String)
   bij System.Configuration.ClientConfigurationSystem.System.Configuration.Internal.IInternalConfigSystem.GetSection(System.String)
   bij System.Configuration.ConfigurationManager.GetSection(System.String)
   bij System.Configuration.ClientSettingsStore.ReadSettings(System.String, Boolean)
   bij System.Configuration.LocalFileSettingsProvider.GetPropertyValues(System.Configuration.SettingsContext, System.Configuration.SettingsPropertyCollection)
   bij System.Configuration.SettingsBase.GetPropertiesFromProvider(System.Configuration.SettingsProvider)
   bij System.Configuration.SettingsBase.GetPropertyValueByName(System.String)
   bij System.Configuration.SettingsBase.get_Item(System.String)
   bij System.Configuration.ApplicationSettingsBase.GetPropertyValue(System.String)
   bij System.Configuration.ApplicationSettingsBase.get_Item(System.String)
   bij QMK_Toolbox.Properties.Settings.get_keymap()
   bij QMK_Toolbox.MainWindow.InitializeComponent()
   bij QMK_Toolbox.MainWindow..ctor()
   bij QMK_Toolbox.MainWindow..ctor(System.String)
   bij QMK_Toolbox.Program.Main(System.String[])
``` 
sorry for the dutch in the logs, could find a quick change of language. However I would say most event log stuff is formatted the same. and "bij" == "at" probably is all you need ;-) 

please note that the regular non-installer version does work.

fauxpark, 2021-01-16:
Try deleting the `C:\Users\<username>\AppData\Local\QMK` directory.

timovp, 2021-01-16:
worked like a charm!! thanks 👍

closed 2021-01-16 by timovp