qmk/qmk_toolbox #285: No DFU capable USB device available - gmmk pro

pierrepgithub, 2021-07-21:
Not sure what the problem is, I'm trying to flash my compiled .bin file to the board and QMK Toolbox is saying "No DFU capable USB device available"..

The same exact steps work on another PC.
I'm using an ANSI gmmk pro
I pasted the terminal:

*** QMK Toolbox 0.0.21 (https://qmk.fm/toolbox)
    Supported bootloaders:
     - Atmel/LUFA/QMK DFU via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via Teensy Loader (https://pjrc.com/teensy/loader_cli.html)
     - ARM DFU (STM32, APM32, Kiibohd, STM32duino) via dfu-util (http://dfu-util.sourceforge.net/)
     - Atmel SAM-BA (Massdrop) via Massdrop Loader (https://github.com/massdrop/mdloader)
     - BootloadHID (Atmel, PS2AVRGB) via bootloadHID (https://www.obdev.at/products/vusb/bootloadhid.html)
    Supported ISP flashers:
     - USBTiny (AVR Pocket)
     - AVRISP (Arduino ISP)
     - USBasp (AVR ISP)
*** STM32 DFU device connected (WinUSB): STMicroelectronics STM32  BOOTLOADER (0483:DF11:2200)
*** Attempting to flash, please don't remove device
>>> dfu-util.exe -a 0 -d 0483:DF11 -s 0x08000000:leave -D "X:\qmk_firmware\gmmk_pro_name.bin"
    dfu-util 0.9
    
    Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2016 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
    
    No DFU capable USB device available

fauxpark, 2021-07-21:
Could you try reinstalling the WinUSB driver? https://docs.qmk.fm/#/driver_installation_zadig?id=uninstallation

First get rid of it and make sure the Toolbox detects it as `NO DRIVER`. Then right click an empty spot in the window and hit `Install Drivers`. Restart the Toolbox, and if it doesn't work, reboot your PC. If *that* doesn't work, repeat the process but this time try the WinUSB driver with Zadig.

pierrepgithub, 2021-07-22:
> Could you try reinstalling the WinUSB driver? https://docs.qmk.fm/#/driver_installation_zadig?id=uninstallation
>
> First get rid of it and make sure the Toolbox detects it as `NO DRIVER`. Then right [...]

My Zadig shows the samething as that document does
HidUsb -> WinUSB

I tried removing the entries in the device manager, but I'm not sure what you mean by :
> and make sure the Toolbox detects it as NO DRIVER

I clicked Scan for hardware changes and windows reinstalled the drivers. But same issue in QMK Toolbox

Could this be an issue with my anti-virus? I use kaspersky.

fauxpark, 2021-07-22:
> My Zadig shows the samething as that document does
> HidUsb -> WinUSB

You're looking at the wrong device and the wrong section of that page, it should have the same name shown in the Toolbox. Hit Options->List all devices:
[image]

> I'm not sure what you mean by :

I mean this should say `NO DRIVER` instead of WinUSB:
[image]

pierrepgithub, 2021-07-22:
This is what my Zadig shows when I plug in my gmmk pro in bootloader mode
[image]

To get that NO DRIVER, should I plug in the board in bootloader mode and delete the STM32 driver in device manager?

fauxpark, 2021-07-22:
Try pressing the Reinstall driver button in Zadig first.

pierrepgithub, 2021-07-22:
> 
> 
> Try pressing the Reinstall driver button in Zadig first.

I plugged the board in into bootloader mode, pressed reinstall driver in Zadig, it succeeded. I restarted the PC, opened toolbox, and same error.

[image]

fauxpark, 2021-07-22:
Could you try looking in Event Viewer to see if there's any errors or warnings that look related?

pierrepgithub, 2021-07-22:
> 
> 
> Could you try looking in Event Viewer to see if there's any errors or warnings that look related?

Can you direct me where in Event Viewer to look? I haven't used this in a long time

pierrepgithub, 2021-07-22:
[image]

There's this?

fauxpark, 2021-07-22:
I found this thread: https://social.technet.microsoft.com/Forums/Lync/en-US/47157c5f-2073-40cc-a648-0ed2ca917b11/event-id-0-detected-unrecognized-driver-warning?forum=win10itprohardware

Which seems to indicate it has something to do with VMWare software. Do you have something like that installed, and is it not installed on the other machine?

pierrepgithub, 2021-07-22:
Yea I do have VMware on the issu PC while the working PC doesn't. Want me to uninstall it?

fauxpark, 2021-07-22:
It seems disabling the VMWare USB arbitration service might fix it, if you're not OK with completely uninstalling. Maybe there is something you can do within the VMWare settings as well, I'm not sure.

pierrepgithub, 2021-07-22:
I was using VMware for some of my InfoSec work. I don't have a problem with either approach if it means progress. I'll look into in the morning, I already signed off for the day. I'll report back after I make those changes. Thank you so much for sticking with me!

pierrepgithub, 2021-07-22:
No go. I tried stopping that service first, same error in Toolbox. Then I uninstalled VMware and restarted - same error in toolbox. :(

fauxpark, 2021-07-22:
Then I'm not really sure what it could be, sorry. Google's not giving me much else.

pierrepgithub, 2021-07-22:
Yea everything looks correct, it just doesn't like me. I haven't reformatted my PC in awhile and I've been meaning to do it. Maybe I'll see if anything changes after that

pierrepgithub, 2021-07-23:
So even after running a "reset PC" on windows 10, I chose to clear all the data and wipe the drives, etc. Still the same error. So weird.

Is it possible to flash it using QMK MSYS?

pierrepgithub, 2021-07-23:
It's crazy to me how this isn't working

fauxpark, 2021-07-23:
Hm, have you tried a different port? And a different cable, if it's not the same one you used on the working PC?

pierrepgithub, 2021-07-23:
It's the same cable, I use a 4 port USB switch that connects to both PCs and I just press the switch button to switch between the PCs

fauxpark, 2021-07-23:
Could you try connecting the board directly? If that doesn't work, I'd say do a proper, real reformat rather than the "Reset PC".

pierrepgithub, 2021-07-23:
my first choice was to do a proper format with the usb, but I wanted to try that method instead. I'll hook up the kb directly to the pc and see

pierrepgithub, 2021-07-23:
Wow, it worked haha. Hmm... so the hub is the issue? Wonder why it still works going through one PC and not the other.. maybe I need to unplug the hub and try reinstalling the drivers that way?

fauxpark, 2021-07-23:
It could be the port that the hub itself is plugged into, but that's just my guess.

pierrepgithub, 2021-07-23:
Yea I'm thinking the port that I'm using to plug the hub into the PC maybe doesnt have enough bandwidth or something? I dunno... Now that we kind of isolated the reason, do you have any suggestions how I can validate or compare the two USB ports on each PC and see if the PC that doesnt work with the hub is somehow incompatible?

pierrepgithub, 2021-07-23:
I changed my xhci controller in the bios to smart auto and I also changed the usb port that the hub plugs into. I was able to flash it now. thanks!

One off topic question if it's ok - is there a way to validate or verify if I'm using the develop branch in QMK MSYS? I did qmk setup -b develop

Just wanted to know if there was a way to verify it's using that branch..

Otherwise, we can close the issue

fauxpark, 2021-07-23:
Good to hear! AFAIK we (QMK) still don't really know what the problem is with hubs, but it does crop up every so often...

If you are in fact on the develop branch, `qmk doctor` will tell you so (in fact this is a new addition to the doctor that is currently only on the develop branch). Or you can `cd` to the `qmk_firmware` directory and run `git status`.

pierrepgithub, 2021-07-23:
I get this, I remember before seeing the word develop or development in msys, not seeing it now. Do I have to setup git for it to work?
`$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.2.0
Ψ QMK home: C:/qmk/qmk_firmware
Ψ Detected Windows 10 (10.0.19041).
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
$`

I don't have git cli installed, is it a requirement?

fauxpark, 2021-07-23:
Git will already be installed inside QMK MSYS. Your CLI version is slightly out of date, and if you are indeed on the develop branch it is also not current.

Run `python3 -m pip install -U qmk`, then:

```
cd ~/qmk_firmware
git status
```

pierrepgithub, 2021-07-23:
I ran the python command, and it said it installed 0.2.2 and uninstalled 0.2.0

But now I get this: 

`$ qmk setup --help
C:/qmk/QMK_MSYS/mingw64/bin/qmk.exe: error while loading shared libraries: MSVCR120.dll: cannot open shared object file: No such file or directory[pierr@DESKTOP-85KIHGA ~]$ qmk doctor
C:/qmk/QMK_MSYS/mingw64/bin/qmk.exe: error while loading shared libraries: MSVCR120.dll: cannot open shared object file: No such file or directory$`

pierrepgithub, 2021-07-24:
i reinstalled qmk msys and same thing

closed 2021-08-04 by pierrepgithub

TunTunMeow, 2021-12-03:
I had the same issue what you have gone through.
Solved it through plugging it into different USB port.
Previous one was usb3.1 port, and tried plugging it into other USB port which is not 3.1 and it worked.