qmk/qmk_toolbox #290: Cannot clear EEPROM on Atmega32u4, and some more EEPROM Issues

4li77l3n00b, 2021-07-26:
[image]
There is a bootloader on my atm chip (according to previous issues), since I can normally flash firmware and use the keyboard.    
My keyboard is a split one which uses EE_HANDS for handedness detection. I've tried to use the following command to manually  
flash eeprom-[left/right]hand.eep (and also reset.eep) files without validation:
`.\dfu-programmer.exe atmega32u4 flash --force --eeprom "eeprom-lefthand.eep" --suppress-validation`
It worked for handedness decision on both sides, but the RGB Matrix on the keyboard stopped working. All RGB bulbs remained red and static(I have enabled them in the firmware for sure). Since the RGB Matrix system makes use of the EEPROM, I guess it was the forcible flashing operation that generated this issue.
## System Information

 - Operating system: Windows 10 Enterprise 21H1
 - QMK Toolbox version: 0.0.21

## Additional Context

fauxpark, 2021-07-26:
The problem in the screenshot has already been fixed in `master` and will be available in the next version - the dfu-programmer command now includes the `--suppress-validation` flag.

The issue with setting handedness resetting the RGB is normal and expected - those `.eep` files essentially set the first 14 bytes of EEPROM to 0, then the 15th byte to either 0 or 1 depending on the handedness you want to set:
[image]
But this also means the "magic number" in the first two bytes of EEPROM is reset to `0x0000`.
[image]

When the firmware next runs, it will see this as an invalid value, and call the function that resets everything to default, including RGB Matrix if the feature is enabled, and the default state of that is static red. All you have to do is use the RGB keycodes to change the colour and effect back to what you had before.

4li77l3n00b, 2021-07-27:
> The problem in the screenshot has already been fixed in `master` and will be available in the next version - the dfu-programmer command now includes the `--suppress-validation` flag.
>
> The issue with [...]

Thanks for that. Now it works.

closed 2021-07-27 by 4li77l3n00b