qmk/qmk_toolbox #305: QMK toolbox flash reset to default

nckeller06, 2021-09-28:
I've been attempting to flash my Massdrop Control keyboard with a new layout using a variety of methods and haven't had any success thus far. Tried using Massdrop's own online config tool and command prompt/terminal on both mac and windows and have spent the last several days trying to figure out QMK configurator and tool box. I've been able to flash my board but every time I get the same error where it can't find the .bin/.hex file despite me opening the correct bin file before flashing. Then it just reflashes the board with the default layout. Wondering if anyone else has had this issue/if there are any potential fixes. Like I said I've looked pretty much everywhere and haven't had any luck but I'm very new to this so I'm sure it's just a mistake I'm making on my end. Thanks in advance for any help

fauxpark, 2021-09-28:
Please provide more information. A screenshot of the Toolbox would help.

nckeller06, 2021-09-28:
Here is the bin file I wish to use in the local file menu
[image]

Then, I select my keyboard and MCU and press load
[image]

If I proceed to flash from here it just flashes the default, which has replaced my bin file at the top

fauxpark, 2021-09-28:
Ignore that whole section, it's for downloading the precompiled default keymap from qmk.fm. In fact, because of this confusion, that will be removed in the next version (the list isn't being updated anyway).

nckeller06, 2021-09-28:
Gotcha. So here is a screenshot of what happens when I flash the keyboard. What I think is happening here is that it's just re-flashing the default bin, meaning my new keymap and changes are not applied
[image]

fauxpark, 2021-09-28:
Yes, that's what the "Keyboard from qmk.fm" section is for - it downloads the firmware and sets the file path to it. As I said, just ignore it.

nckeller06, 2021-09-28:
Understood. So then my question is how can I actually flash my new firmware to the board?

fauxpark, 2021-09-28:
You ignore the "Keyboard from qmk.fm" section.

nckeller06, 2021-09-28:
Oh alright, I'll give that a try real quick and get back to you. Thank you so much for the help!

nckeller06, 2021-09-28:
Well, that seems to have mostly worked. The copy and paste keys are still not working, but that might be something I did wrong in the configurator. The media mute/prev/next keys seem to be working tho

fauxpark, 2021-09-28:
If you used `KC_COPY`/`KC_PASTE`, then they won't work unless you're on Linux: https://docs.qmk.fm/#/keycodes?id=basic-keycodes

nckeller06, 2021-09-28:
Well that would explain that! Is there any way I can get it to work for Windows or Mac?

fauxpark, 2021-09-28:
Just use Ctrl+C/Ctrl+V.

nckeller06, 2021-09-28:
But is there a way I can bind those two inputs to one key? To set the keymap I've been using QMK configurator so I'm not sure how to do that

nckeller06, 2021-09-28:
And I apologize because I know this is all super basic stuff but I'm just getting into  mechanical keyboard building so I'm trying to figure it all out

fauxpark, 2021-09-28:
Yes, it's under the Quantum tab.

nckeller06, 2021-09-28:
Thank you, I will give that a try

nckeller06, 2021-09-28:
That worked!! My final and much less pressing question is about RGB. Is there any way to program keyboard RGB using QMK?

fauxpark, 2021-09-28:
That's a pretty broad question, it depends what you mean by "program" and "QMK", but generally the answer is "yes".

nckeller06, 2021-09-28:
I appreciate the use of quotation marks. Like I said, I don't know what I'm doing, nor do I know the correct terminology. Thank you for your help

closed 2022-04-11 by nooges