qmk/qmk_toolbox #322: "Install Drivers..." Not Working On Windows 11

oxmhpark, 2021-12-21:
1. Installed QMK_MSYS.exe
2. Installed qmk_toolbox_install.exe
3. Run QMK Toolbox as administrator
4. Run QMK Toolbox > Tools > Install Drivers...
5. Every installation steps start with "Extracting driver files... Success" but end with "Installing driver... This can take a while, please, don't close this window! Other error" line
6. Connect DZ60 keyboard with DFU mode
7. QMK Toolbox shows message: "*** Atmel DFU device connected (NO DRIVER):  ATm32U4DFU (03EB:2FF4:0000)"
8. Tried some .hex flashing; dz60_default.hex, dz60_test.hex, ...
8. Flashing is not successful.

paul-omalley, 2022-01-03:
Yes, this has been an issue ever since the initial Insider release of Windows 11.
Basically, the issue is that the Driver Installer can't make the changes it needs to install the drivers. 

I sure hope the team can fix this soon, especially with a lot of people moving to Windows 11 and expecting their software to work.

zerodividesbyyou, 2022-02-09:
Hey guys, just ran into this issue. The fix for me was to disable driver integrity checks after also disabling Secure Boot in my UEFI/BIOS. In an elevated command prompt (once secure boot is disabled) run "bcdedit.exe /set nointegritychecks on" and driver signing will be disabled and all the drivers should yield success in the QMK Installer. Afterwards, run the same command with "off" at the end instead of on. Voila! Fixed

mautrok, 2023-09-20:
@zerodividesbyyou, tried your solution this way:
- disabled secure boot driver integrity chek
- start powershell as admin run the command above
- from qmk toolbox launched   "qmk compile -kb lets_split -km default"

but i still have same issue the keyboard is not recognized yet, same keyboard on other OS works.

Anyone has any updates about this issue?