qmk/qmk_toolbox #358: QMK toolbox 0.2.0 does not recognize devices (macOS)

DarrienG, 2022-04-29:
## Describe the Bug

QMK 0.2.0 does not recognize devices. If I enter reset mode or DFU mode on my keyboards QMK no longer recognizes them. I have a keebio Iris rev 5 and an Iris rev 6a. In both cases entering reset or DFU mode nothing shows up on QMK 0.2.0. 

When I plug in my rev 6, it does show up as an HID device, but no longer recognizes it when I enter reset mode or DFU mode.

Using the old 0.1.1 build fixes the issue and everything works as expected.

## System Information

 - Operating system: 
Running on an M1 MacBook Air.

```
🜛 keymaps [master●] uname -a
Darwin dg-air 21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:47:26 PDT 2022; root:xnu-8020.101.4~15/RELEASE_ARM64_T8101 arm64
```
[image]

 - Reproduceable only with QMK toolbox version: 0.2.0

## Additional Context

As mitigation steps I tried a number of things:
* Plugging in only one half of my keyboard
* Using a couple of different USB C cables (one with USB 2, one with USB 3)
* Plugging directly into the device rather than using a hub

But couldn't get it to work at all.

Installed with:

```
$ brew tap homebrew/cask-drivers
$ brew install --cask qmk-toolbox
```

I did a fresh reinstall just in case something about the install was corrupted, but that didn't fix it either.

fauxpark, 2022-04-29:
Fixed by https://github.com/qmk/qmk_toolbox/pull/350

DarrienG, 2022-04-29:
Wow you guys are fast :0

DarrienG, 2022-04-29:
Will close this out, thanks for the fast comment :)

closed 2022-04-29 by DarrienG