qmk/qmk_toolbox #366: Search keyboards on qmk.fm

YassineGherbi, 2022-05-14:
I am not sure when or really why this has been removed, the main reason I could find is user error and spaghetti code. But I would like to make an effort to return or reimplement the search bar.  For a lot of beginners, this is the main reason to use QMK Toolbox.

I looked around youtube where most beginners will look for tutorials and in every single video, they rely on this functionality.

## Feature Request Type

- [x] Core functionality
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

If I am missing something please let me know. I personally don't see the value of Toolbox if the user has to touch the cli anyway to build a hex file.

fauxpark, 2022-05-14:
This feature relied on Travis CI to upload default keymap firmware for each board in the repo. It has since been shut down and so the list is not updated anymore. Therefore it has been removed.

closed 2022-05-14 by fauxpark

YassineGherbi, 2022-05-14:
This is for many the main reason to use Toolbox. Please take a look at any blog of video tutorial. They all rely on this. 

This will set a high barrier for people trying to get into this hobby. 

I understand what you are saying but is there no way we could make this work without the use of Travis CI.

YassineGherbi, 2022-05-14:
Maybe there’s a way to achieve this using the github api? I didn’t look in to the code yet to see what data is needed but I can’t imagine this not being possible without the use of TravisCI.

nooges, 2022-05-14:
This feature actually caused quite a bit of confusion for users loading in their own .hex/.bin file generated from [QMK Configurator](https://config.qmk.fm). We would frequently see people opening up their own downloaded file, but then select their keyboard from the qmk.fm load section of Toolbox, and flash that file to their board instead.

CLI is not the only way to get a the firmware file, as we have [QMK Configurator](https://config.qmk.fm) and the firmware files at https://qmk.fm/keyboards/ and https://qmk.tzarc.io/

This feature also had the limited use case of only downloading the default keymap for the keyboard, so QMK Configurator is the suggested method for beginners as mentioned here: https://docs.qmk.fm/#/?id=what-is-qmk-firmware