qmk/qmk_toolbox #373: error opeing hex files.

Angelsanche, 2022-07-02:
So iwas trying to flash my ginko65 with the hex files that is available on the QMK configurator and at first the keys were messed up so I reset and now tried another hex file and now my keyboard wont type any keys and everytime I try to flash it it says 
> Error opening C:\Users\derpa\Downloads\mokey_ginkgo65hot_layout_mine.hex
> See --debug=51 or greater for more information.
And I'm not really sure how to fix it.

fauxpark, 2022-07-03:
Could you please show the full output?

Angelsanche, 2022-07-03:
> Could you please show the full output?
Attempting to flash, please don't remove device
> dfu-programmer.exe atmega32u4 erase --force
> Erasing flash...  Success
> Checking memory from 0x0 to 0x6FFF...  Empty.
> dfu-programmer.exe atmega32u4 flash --force "C:\Users\derpa\Downloads\mokey_ginkgo65hot_layout_mine.hex"
> Error opening C:\Users\derpa\Downloads\mokey_ginkgo65hot_layout_mine.hex
> See --debug=51 or greater for more information.
> dfu-programmer.exe atmega32u4 reset
Flash complete

fauxpark, 2022-07-03:
Does that file exist? If so, try redownloading it from the Configurator, it may be corrupted somehow.

Angelsanche, 2022-07-03:
> 

So redownloaded and now im able to type but I think the hex file in the QMK configurator is broken or something. For some reason the windows key types 1 and the ctrl keys types 2.

fauxpark, 2022-07-03:
Are you sure you're flashing the correct firmware for the board you have?

Angelsanche, 2022-07-03:
> Are you sure you're flashing the correct firmware for the board you have?

yes it is the mokey ginkgo65, I also think that people were having the same issues of the firmware being bugged and the ctrl being the escp button and the windows being the 1 button but I'm not really sure how to return the keys back to normal.

fauxpark, 2022-07-03:
Try clearing the EEPROM with the Toolbox and reflashing. I've looked at the code for this board and it appears okay, but perhaps the pinout and/or matrix is incorrect which would not be immediately obvious.

Angelsanche, 2022-07-03:
> Try clearing the EEPROM with the Toolbox and reflashing. I've looked at the code for this board and it appears okay, but perhaps the pinout and/or matrix is incorrect which would not be immediately obvious.

https://github.com/qmk/qmk_firmware/issues/17307#issuecomment-1172685208
Seems like people may have figured it out in this thread but I'm really new to this qmk thing and I'm not really sure how to fix it. It would be greatly appreciated if you could help me.

fauxpark, 2022-09-26:
@Angelsanche seems like the PR linked in that issue has been merged, can this be closed or are you still having problems with this board?