qmk/qmk_toolbox #374: RP2040 suppord

HughMungis, 2022-07-03:
## Feature Request Type

- [ ] Core functionality
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

now that QMK supports the RP2040 board family, can we get that included in the toolbox?

fauxpark, 2022-07-03:
I don't really see a need, you can just drag and drop the firmware into the drive.

HughMungis, 2022-07-03:
> I don't really see a need, you can just drag and drop the firmware into the drive.

Is that the case for QMK? I know for KMK, the drag and drop of the config files is how the board gets the instructions it uses to run. I was under the assumption that I would need to flash QMK firmware onto the board itself, but I admit this is a knowledge gap for me.

fauxpark, 2022-07-03:
It doesn't matter what the firmware is, QMK, KMK, or anything else. The RP2040 itself comes with a UF2-compatible bootloader that appears as a flash drive. All you need to do is drag the .uf2 file onto it.

closed 2022-07-12 by HughMungis