qmk/qmk_toolbox #378: Arm binaries for dfu-programmer etc.

Muirium, 2022-07-16:
The command line utilities QMK Toolbox uses internally are currently all Intel builds on Mac. Run file * on the Resources directory inside the app to check them out. This is not a problem yet, as they run under Rosetta. But fully native builds are a good idea. I substituted dfu-programmer for the Apple Silicon native version in [Homebrew](https://formulae.brew.sh/formula/dfu-programmer#default). Works perfectly. But now I've updated QMK Toolbox to the current version, I see they are Intel only again.

## Feature Request Type

- [ ] Core functionality
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

fauxpark, 2022-07-16:
The Toolbox itself is currently built as a universal binary, but the deployment target is set to 10.15 (following Homebrew minimum system requirements because the flasher binaries are taken from the bottles). Even then, I don't think Brew builds universal binaries - the bottles are separate for Intel and Apple Silicon.

Muirium, 2022-07-16:
True!

`dfu-programmer:     Mach-O 64-bit executable arm64`
`dfu-util:           Mach-O 64-bit executable x86_64`

I’m not technical to know how hard it is to “stitch” both architectures into a single Universal binary. Hadn’t even realised my build was exclusively Arm64. 

Obviously enough, Universal is the way to go. Plenty of Intel Macs still out there. For now…

joxelstudios, 2024-02-28:
Commenting here instead of creating a new issue. Just tested the latest release ([0.3.0](https://github.com/qmk/qmk_toolbox/releases/tag/0.3.0)) and can confirm the intel DFU binaries run and flash fine under Rosetta. I flashed an update for my Nuphy Air60 V2 on an M1 Max Macbook Pro.

jajajaime, 2024-07-11:
Adding a comment as well that using the latest release 0.3.3 did not give me any issues on a MacBook Pro M1 Pro for a Nuphy Air75 V2.

Muirium, 2025-06-20:
Apple has now announced the end of life for Rosetta 2:

https://arstechnica.com/gadgets/2025/06/apple-details-the-end-of-intel-mac-support-and-a-phaseout-for-rosetta-2

The current Intel binaries won’t work on macOS 28 in 2027. I substituted arm binaries installed in Homebrew on my own Mac, and recommend using these in future releases of QMK Toolbox. Intel is obsolete and those Macs are a dwindling user base.