qmk/qmk_toolbox #381: Application not running

AlessandroMinoccheri, 2022-10-04:
Hi all, I am trying for the first time to use my split keyboard so I create a custom layout, downloaded the .hex file and download QMK toolbox app.

I started the application open the hex file but nothing happens.

In the bottom line I can see No HID console devices connected.

I ma suing an ergodox 84 characters.
OS> Mac BigSur 11.3.1

How can I use my custom layout?

keyboard-magpie, 2022-10-04:
To flash using QMK toolbox, you will need to put your keyboard into bootloader mode. Usually there is a separate reset button that should be pressed with the keyboard plugged in.

Once the bootloader is recognised, you should see a line of yellow text in the Toolbox console, you can then click "flash".

AlessandroMinoccheri, 2022-10-04:
Ok @keyboard-magpie now I can see this yellow line>

`HalfKay device connected: (null) (null) (16C0:0478:0120)
`

keyboard-magpie, 2022-10-04:
> you can then click "flash".

closed 2022-10-04 by keyboard-magpie

AlessandroMinoccheri, 2022-10-04:
I got it!!! I have not found these steps easily it could be useful to be more explicit what do you think?

keyboard-magpie, 2022-10-04:
https://docs.qmk.fm/#/newbs_flashing?id=flashing-your-keyboard-with-qmk-toolbox