qmk/qmk_toolbox #383: .NET/QMK error not starting properly nor detecting devices

MackanT, 2022-10-26:
## Describe the Bug

After using QMK Toolbox for 1-2 years it stopped working, in that it would neither boot nor show up in task manager (started both normally and as administrator). Since it didn't work I thought that a quick fix would be to uninstall it and install the current 0.2.2 version (I also tried a few other versions: 0.2.1, 0.2.0 and the current beta without success) however upon finishing the installation I'm greeted by the below error message, and pressing details gave the long error message. 

I can press the continue button to get into the app but it does not seem to work properly as it neither detects my previously built keyboards nor new out-of-the-box Pro-Micros. I also went ahead and downloaded the drivers but it still won't detect nor allow me to flash anything

It seems to be some miss between .NET framework and QMK, but I'm far from an expert at reading these types of error messages.... 
Sorry if this is the wrong place to post this, but I couldn't find anything as either an old issue or a common .NET issue on google!

## System Information

 - Operating system: Windows 10 Home, Version: 21H2, Build: 19044.2130
 - QMK Toolbox version: 0.2.0, 0.2.1, 0.2.2

## Additional Context

[image]

[image]

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.Management.ManagementException: Invalid class 
   at System.Management.ManagementException.ThrowWithExtendedInfo(ManagementStatus errorCode)
   at System.Management.ManagementObjectCollection.ManagementObjectEnumerator.MoveNext()
   at System.Linq.Enumerable.<CastIterator>d__97`1.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   at QMK_Toolbox.Usb.UsbListener.EnumerateUsbDevices(Boolean connected)
   at QMK_Toolbox.Usb.UsbListener.Start()
   at QMK_Toolbox.MainWindow.MainWindow_Load(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at QMK_Toolbox.MainWindow.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4515.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
qmk_toolbox
    Assembly Version: 0.2.2.0
    Win32 Version: 0.2.2
    CodeBase: file:///C:/Program%20Files/QMK%20Toolbox/qmk_toolbox.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4550.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4536.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4190.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4566.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
Microsoft.GeneratedCode
    Assembly Version: 1.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Deployment
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4270.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Deployment/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Deployment.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4435.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
[image]

fauxpark, 2022-10-27:
I'm not able to reproduce this on a fresh 21H2 Home VM. Makes sense as .NET Framework 4.8 is preinstalled since May 2019: https://learn.microsoft.com/en-us/dotnet/framework/migration-guide/versions-and-dependencies#net-framework-48

The stack trace unfortunately isn't very helpful. The error is happening somewhere in the WMI code; something is probably wrong with your OS install. What that is, impossible to tell...

Only thing I can say is make sure everything is up to date.

fauxpark, 2022-10-27:
I did find this: https://kb.eway-crm.com/faq-1/installation/sql-server-cannot-be-installed-due-to-wmi-error
However the WMIDiag link is dead, and on further research it seems it's no longer supported.
The other thing it says is
> The recommended solution is to reinstall the operating system

Fun.

MackanT, 2022-10-27:
Mmmmmm thanks for the help! 
But that does not sound like a fun solution xD

MackanT, 2022-10-30:
I at least got the flashing working from the terminal, so all should be OK

fauxpark, 2022-12-17:
@MackanT if you're still having this problem, could you please try the .exe from the above PR?

MackanT, 2022-12-17:
I tried checking out the PR but I cannot find a .exe in the file, and I do not really know how to build it myself. 
I can see that in the automatic check of the PR, it builds an qmk_toolbox_install.exe and qmk_toolbox.exe, but I cannot find any of them in the file structure itself

fauxpark, 2022-12-17:
The .exe is not part of the source code, it's a build artifact. You can just download it from there.

MackanT, 2022-12-17:
Okay, I found the downloads now, thanks! :)  Tried both with the installer and without, but unfortunately still got a error message that looks the same: 

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.Management.ManagementException: Invalid class 
   at System.Management.ManagementException.ThrowWithExtendedInfo(ManagementStatus errorCode)
   at System.Management.ManagementObjectCollection.ManagementObjectEnumerator.MoveNext()
   at System.Linq.Enumerable.<CastIterator>d__97`1.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   at QMK_Toolbox.Usb.UsbListener.EnumerateUsbDevices(Boolean connected)
   at QMK_Toolbox.Usb.UsbListener.Start()
   at QMK_Toolbox.MainWindow.MainWindow_Load(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at QMK_Toolbox.MainWindow.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4515.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
qmk_toolbox
    Assembly Version: 0.2.2.0
    Win32 Version: 0.2.2
    CodeBase: file:///C:/Program%20Files/QMK%20Toolbox/qmk_toolbox.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4550.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4536.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4190.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4590.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
Microsoft.GeneratedCode
    Assembly Version: 1.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Deployment
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4270.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Deployment/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Deployment.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4435.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

MackanT, 2022-12-17:
Accidentally marked it as closed. I don't know if you want it closed or opened, so I'll reopen it until I hear anything else.

fauxpark, 2022-12-17:
I've found this article, might be helpful?
https://techcommunity.microsoft.com/t5/ask-the-performance-team/wmi-missing-or-failing-wmi-providers-or-invalid-wmi-class/ba-p/375485

MackanT, 2022-12-18:
Thanks for the help! The link in itself did not fix the issue, but it led me in the right direction!
Apparently, it was a windows register error all along which caused this specific app (QMK Toolbox) to not function while all other programs worked correctly. If anyone else has a similar issue, the following link with info helped solve it for me: https://learn.microsoft.com/en-us/answers/questions/684166/failed-to-initialize-all-required-wmi-classes.html

closed 2022-12-18 by MackanT

Mast3rYoda, 2023-04-17:
Thanks @MackanT  I had the exact same issue and your provided link solved it!

bbarber9, 2024-02-02:
@MackanT can you share what exactly you followed from that link?

MackanT, 2024-02-03:
I unfortunately don't remember 100%, but I believe it was the top-rated answer by Limitless Technology

PR #395 (2023-03-16): [Windows] Do not prefer 32-bit

## Description

On 64-bit Windows the Toolbox should run as a 64-bit application.

It's possible this fixes #383. Will need testing.

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

*