qmk/qmk_toolbox #384: Unable to force flash firmware onto PCB

boogboards, 2022-10-27:
Hi, I tried to reflash a keyboard with a firmware but now the keyboard itself is completely unresponsive and won't exit bootloader mode.

This is the issue as described, not sure how to rectify.

Atmel DFU device connected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)
Attempting to flash, please don't remove device
> dfu-programmer.exe atmega32u4 erase --force
> Erasing flash...  Success
> Checking memory from 0x0 to 0x6FFF...  Empty.
> dfu-programmer.exe atmega32u4 flash --force "C:\Users\User\Downloads\gopolar_gg86_via.hex"
> Bootloader and code overlap.
> Use --suppress-bootloader-mem to ignore
> dfu-programmer.exe atmega32u4 reset
Flash complete

appreciate any and all help!

sigprof, 2022-10-27:
Looks like the `gopolar_gg86_via.hex` firmware that's available from https://www.caniusevia.com/docs/download_firmware is actually too large and does not fit into flash together with the `atmel-dfu` bootloader. I'm not sure how the VIA maintainers were able to build that firmware (maybe they disabled the size check somehow).

You can try using another firmware build instead: https://raw.githubusercontent.com/Xelus22/QMK-VIA-Hex/master/gopolar/gopolar_gg86_via.hex (this one is a nightly build from the `master` branch; it will probably open as text in the browser — use the “Save As…” command then)

keyboard-magpie, 2023-12-13:
Closed as completed

closed 2023-12-13 by keyboard-magpie