qmk/qmk_toolbox #386: Qmk Toolbox unable to open uf2 files

dissmaterial, 2022-11-09:
I have been trying to flash an STM32 using the newest version of qmktoolbox 0.2.2. I have been able to successfully connect the bootloaded STM32, however qmk toolbox does not let me open uf2 files - even though it is stated to be supported. Instead it will only allow me to open .bin or .hex files

help would be much appreciated

keyboard-magpie, 2022-11-09:
If you truly have an stm32 that takes UF2 it should present as a mass storage device that you can drag and drop UF2 files into, without requiring Toolbox.

fauxpark, 2022-11-09:
The Toolbox does not support UF2 files. I'm not sure where you saw that.

dissmaterial, 2022-11-09:
ahh - all working now - thanks for the point in the right direction!:

closed 2022-11-09 by keyboard-magpie