qmk/qmk_toolbox #391: Virus Total hits

williamdean19, 2022-12-04:
Getting the four warnings on virustotal for the windows installer .exe file:

I saw another issue about the code not being signed. I would like to see this signed to feel safer about using it.
You can close this issue if it's too related to the other one.

Thanks,
Will

CrowdStrike Falcon
Win/grayware_confidence_70% (D)

Cynet
Malicious (score: 100)

Tencent
Trojan.Win32.Offloader.zk

Trapmine
Suspicious.low.ml.score

[image]

keyboard-magpie, 2022-12-04:
These are expected hits when using this software. 

Signing is happening, review the current PRs.

closed 2022-12-04 by keyboard-magpie

tzarc, 2022-12-04:
[image]

Signed build.

williamdean19, 2022-12-06:
Thanks for the follow up! Sorry if I missed a posted update. Thanks,
Will