qmk/qmk_toolbox #397: No DFU capable USB device available

Millhaws, 2022-12-30:
I have been trying to flash my GMMK Pro with new firmware but every time  I try, I get the same error codes "Cannot open DFU device 0483:DF11 found on devnum (LIBUSB_ERROR_NOT_SUPPORTED)" and "NO DFU capable USB device available". I'm pretty new to custom keyboards so maybe I'm missing something, or maybe it's an issue with the GMMK Pro itself, but regardless it's pretty annoying that I can't seem to get this to work. 

I have tried multiple different possible solutions but have not made any progress in getting any new firmware onto the board.

## Describe the Bug
<
I've downloaded the latest version of QMK Toolbox, and have tried to flash multiple different firmware with no luck. I tried,
- GMMK Pro firmware from caniusevia.com, both rev1, AND rev2 with no luck
- following this step-by-step tutorial on how to get VIAL to work https://www.describee.com/newbie-guide-to-installing-vial-and-qmk-on-the-gmmk-pro/
- updating the USB Drivers with Zadig
- plugging the board into multiple different USB ports, some on the motherboard of my PC as well as in the USB hub that I use for all of my peripherals
- Asking around online for other solutions, but have not had much success finding any more information thus far

No matter what I do I always get the same errors  "Cannot open DFU device 0483:DF11 found on devnum (LIBUSB_ERROR_NOT_SUPPORTED)" and "NO DFU capable USB device available"
## System Information

 - Operating system: Windows 10
 - QMK Toolbox version: 0.2.2

I have attached a screen shot of the QMK Toolbox with the errors for clarity. I just want to get firmware that's compatible with via or vial so I don't have to use Glorious Core anymore. Any assistance in resolving this issue would be greatly appreciated!

[image]

sigprof, 2022-12-31:
Apparently the mentioned `GuiSTDFUDev` driver is a component of some Guillemot/Thrustmaster software, which has a very annoying behavior of immediately reinstalling that driver whenever it gets replaced, therefore the “Tools → Install drivers” command does not help in this situation.

Some people were able to work around the problem by killing some Thrustmaster processes from the Task Manager, so that the driver reinstallation does not happen immediately (unfortunately, I don't know the exact names of those processes — if you find out, please report your findings here). Another option is uninstalling the offending Thrustmaster software completely.

Millhaws, 2022-12-31:
Great that worked! you were right about the issue being a conflict with my Thrustmaster HOTAS drivers. I just had to uninstall that and the flash worked. Thank you so much for your help. I have been trying every other solution for the past three days so I'm glad that it's finally working! The next step is just to try and learn VIA. Thanks again for your help!!

closed 2022-12-31 by Millhaws