qmk/qmk_toolbox #398: Driver isntall for GMMK2 not working

desultory, 2023-01-15:
When my keyboard is in bootloader mode, it's recognized but it says the drivers are not installed.  I tried installing through this toolbox and restarting, and that did not help.

Following error:
`WB32 DFU device connected (NO DRIVER):  WB Device in DFU Mode (342D:DFA0:0100)`

## Describe the Bug

## System Information

 - Operating system:  Windows 11 22H2 Microsoft Windows [Version 10.0.22621.1105]
 - QMK Toolbox version: 0.2.2

## Additional Context

followed this guide
https://www.gloriousgaming.com/blogs/guides-resources/gmmk-2-qmk-installation-guide

fauxpark, 2023-01-16:
The current stable version of the Toolbox does not support the WB32 bootloader. Support has been merged but I haven't made a new release yet.

desultory, 2023-01-16:
> The current stable version of the Toolbox does not support the WB32 bootloader. Support has been merged but I haven't made a new release yet.

any pointers to where I can find this and maybe make it myself? Do you also know of any method to just manually do this using wsl, powershell, or a native linux host? I would much rather do that.

fauxpark, 2023-01-16:
Flashing instructions for each bootloader are in the docs, but it doesn't have WB32.
https://docs.qmk.fm/#/flashing

The beta version can be found in the releases tab, keep in mind you might have problems with it.

sigprof, 2023-01-16:
One particular caveat is that you will need to clean out the `C:\Users\USERNAME\AppData\Local\QMK\QMK Toolbox` directory, and then run “Tools → Install Drivers” from the beta version.  The problem is that the beta version internally still has the same version number as the release version, and therefore skips extracting the driver installer, unless you remove the old files first.

desultory, 2023-01-16:
> One particular caveat is that you will need to clean out the `C:\Users\USERNAME\AppData\Local\QMK\QMK Toolbox` directory, and then run “Tools → Install Drivers” from the beta version. The problem is [...]

thanks for that bit of info, i'll do that.
Do you also have a build environment that works with wsl? I found this project from the gmmk docs and they are a bit lacking.  Do you know if compiling the binaries with more jobs can break them? I did that when I remade my keymap.

desultory, 2023-01-16:
deleted the whole QMK folder, ran the beta installer, opened it and tried to install drivers, got this:
```
STM32 Bootloader
  Extracting driver files...
    Success
  Installing driver...
    This can take a while, please, don't close this window!
libwdi:error [install_driver_internal] installer failed to respond - aborting
libwdi:error [install_driver_internal] installer failed to shut down - terminating
```
should I try to reboot?

sigprof, 2023-01-16:
Try rebooting; hope it helps (debugging those Windows problems might be really hard).

desultory, 2023-01-16:
> Try rebooting; hope it helps (debugging those Windows problems might be really hard).

yeah that is why i'd prefer to do this on linux if possible

desultory, 2023-01-16:
a reboot fixed it! thank you very much, now my right control works on my gmmk2 :) i'd happily do some testing for this hardware/firmware if needed