qmk/qmk_toolbox #403: Keyboard freezing after 5 press of the lighting mode switch

duckyz1, 2023-02-11:
## Describe the Bug

## System Information

 - Operating system: 
 Windows 10 Pro 21H2 19044.2486
 
 - QMK Toolbox version: 
0.2.2
## Additional Context

Attempting to flash, please don't remove device
> dfu-programmer.exe atmega32u4 erase --force
> Erasing flash...  Success
> Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
> dfu-programmer.exe atmega32u4 flash --force "C:\Users\DuckyZ1\Downloads\mwstudio_mw75_default.hex"
> 0%                            100%  Programming 0x6A00 bytes...
> [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
> 0%                            100%  Reading 0x7000 bytes...
> [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
> Validating...  ERROR
> 27018 invalid bytes in program region, 1530 outside region.
> FAIL
> Memory did not validate. Did you erase?
> dfu-programmer.exe atmega32u4 reset
Flash complete
Atmel DFU device disconnected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)

duckyz1, 2023-02-11:
Im like almost sure this is a corrupted chip of somesort. I can't isp flash I don't have any experience of doing so. I also can't buy. are there any ways on how I can fix this using software???

keyboard-magpie, 2023-02-11:
No. 

I agree this looks like bootloader corruption. The only way to fix this is with an external ISP flasher. This can be something as simple as a pro micro.

tzdhbo21, 2023-03-03:
> Im like almost sure this is a corrupted chip of somesort. I can't isp flash I don't have any experience of doing so. I also can't buy. are there any ways on how I can fix this using software???

Follow up pls. Mine meets the similar issue. Cannot read Cannot flash

duckyz1, 2023-12-13:
bump, any fix? my keyboard has been stuck with no lighting I kind've miss the lights.

tzarc, 2023-12-13:
The “fix” has been described — use ISP flashing to reinstall the bootloader.

closed 2023-12-13 by tzarc

keyboard-magpie, 2023-12-13:
Fix was stated above. If you're having issues come and discuss on the discord server. Otherwise, refer to https://docs.qmk.fm/#/isp_flashing_guide?id=isp-flashing-guide