qmk/qmk_toolbox #405: I can't flash my keyboard since I'm not allowed to press the "flash" button.

Sardinburk, 2023-02-23:
## Describe the Bug
I'm trying to flash my newly built custom keyboard with it's needed firmware but I can't press the "flash" button. The button is faded in gray as I can't click it. 
I've installed the drivers in QMK Toolbox, I have the right firmware file (.hex file) selected and the right MCU (ATmega32U4) selected.
I did some research and if everything works it's supposed to come up yellow text when I connect and disconnect the keyboard but it doesn't. Nothing comes up but when I tick the "Show All Devices" option under the "Tools" tab this comes up when I connect and disconnect the keyboard:

USB device connected (usbccgp): (Standard USB Host Controller) USB Composite Device (1EA7:7777:0111)
USB device connected (HidUsb): (Standard system devices) USB Input Device (1EA7:7777:0111)
USB device connected (HidUsb): (Standard system devices) USB Input Device (1EA7:7777:0111)
USB device connected (HidUsb): (Standard system devices) USB Input Device (1EA7:7777:0111)

I don't know why there's four lines when I connect one device.
I tried to enter DFU mode like this: keyboard unplugged, held spacebar and b then plugged the keyboard in and waited a few seconds then released spacebar and b. 
After I put he keyboard in DFU mode, this comes up:

USB device connected (NO DRIVER):  DFU in FS Mode (2E3C:DF11:0200)

What is "FS mode"?
I guess it's supposed to say that I have a driver.

## System Information

 - Operating system: Windows 11 22H2
 - QMK Toolbox version: 0.2.2

## Additional Context
Don't know if this is a bug or not but I felt like the context of my problem fits as a bug report.

keyboard-magpie, 2023-02-23:
> NO DRIVER

Install the bootloader drivers from the tools menu of QMK toolbox.

Sardinburk, 2023-02-23:
Hello,
I've already done that about 3 times.
I tried it just now and the same message shows:

USB device connected (NO DRIVER):  DFU in FS Mode (2E3C:DF11:0200)

keyboard-magpie, 2023-02-23:
Which keyboard is this?

Sardinburk, 2023-02-23:
It's a Zoom87 also known as Zoom TKL.

keyboard-magpie, 2023-02-23:
If it's an r2 dual wired/wireless PCB it may not be flashable with QMK toolbox (this was the case for meletrix' other wired/wireless PCBs)

Sardinburk, 2023-02-23:
So basically the Zoom87's PCB is "unflashable". I assume that you can only flash via wire and not Bluetooth. Mine is connected via wire.

keyboard-magpie, 2023-02-23:
You should be approaching meletrix/wuque for instructions, they have a guide for their other dual-mode keyboards.

closed 2023-02-23 by keyboard-magpie

Sardinburk, 2023-02-23:
Alright, thank you.