qmk/qmk_toolbox #408: Error reading line 1

Ex0nys, 2023-03-21:
Hi, I’m trying to flash a q1 and I got the error that I wrote above, error reading line 1. I followed some instructions on Reddit for a similar one and compiled some changes to the rgb file “h config” I think it’s called and compiled a new firmware. Then when I try to flash it onto the q1 it reads well but stops immediately because of the error and I can’t seem to find anything about it. Did I do something wrong? Is there a way to fix it?

I was flashing a bin file if that’s relevant

fauxpark, 2023-03-21:
You need a hex file, not a bin file. You have the wrong firmware.

Ex0nys, 2023-03-21:
Okay thanks, I’m really new to this sry for inconvenience. And thanks again <3

Ex0nys, 2023-03-21:
[image]
So in other words this is not the correct file?, but I can’t seem to find the hex file?

fauxpark, 2023-03-21:
Looks like you're using the Keychron fork. You need to use one of the `atmega32u4` versions, not `stm32l432`.

Ex0nys, 2023-03-21:
Aaaaah okay will test again still much love

HaynesDC5, 2023-09-10:
> Looks like you're using the Keychron fork. You need to use one of the `atmega32u4` versions, not `stm32l432`.

Hello, I am having the exact same issue but I am already using "ATmega32U4" Any idea what is causing this? The last try I switched to u2 just to see if it would work.
[image]

fauxpark, 2023-09-10:
You are still using the wrong firmware file. It needs to be a `.hex`, not `.bin`.

HaynesDC5, 2023-09-10:
> You are still using the wrong firmware file. It needs to be a `.hex`, not `.bin`.

Sorry if this is a basic question but how can I convert it? This is the file directly from Keychrons website so I'm not sure what needs to be done here. I'm brand new to this. Thank you for your help in advance.

fauxpark, 2023-09-10:
You can't convert it, you need the right file.

HaynesDC5, 2023-09-10:
> You can't convert it, you need the right file.

Is there anywhere I can obtain it? Ive been searching for hours and haven't seen anything about this listed.

sigprof, 2023-09-10:
@HaynesDC5 Looks like you are trying to flash a firmware for the Keychron **Q1 Pro** board, when you have some other board (maybe Keychron **Q1 V1**, because it seems to be the only relevant Keychron board that uses atmega32u4).

Stock firmware files for Keychron Q1 can be found at https://www.keychron.com/blogs/archived/how-to-reset-your-keychron-q1-to-factory-settings — you need to choose the Version 1 file that matches your board (there are both ANSI and ISO layout variants, and knob vs non-knob variants).

HaynesDC5, 2023-09-10:
> 
Thank you, this combined with the response I just got from Keychron support has made me realize some things. I double checked my order and I did indeed purchase a Q1 PRO V2. Apparently I did get the shell and build of this keyboard but the board of a Q1 V1. I'm assuming this is in part due to the fact that I ordered a limited special edition offer. Tbh I really don't care, it gets the same job done but this entire problem was caused by Keychron having changed the boards on some V2 special editions with V1s. I was able to successfully flash a Q1 V1 file even though I double checked that it shouldn't have worked. I appreciate the help everyone.

sigprof, 2023-09-10:
Ewww, so in addition to Q1 Version 1 and Q1 Version 2 they also have “Keychron Q1 Version 2 Special Edition”, which has a separate firmware on [the Q1 firmware page](https://www.keychron.com/blogs/archived/how-to-reset-your-keychron-q1-to-factory-settings), and apparently the PCB for that version uses atmega32u4, like Q1 Version 1.  No idea whether that firmware has any important differences from the Q1 Version 1 firmware — the Keychron fork at https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/keychron does not seem to have any code specific to that “special edition”.

HaynesDC5, 2023-09-10:
Well after a bunch of struggling its safe to say there is definitely a difference. Layers and macros didn't work at all as far as I could tell. Even after flashing with the V2 SE file its iffy. For some reason Layer 2 is now default and nothing works to reach Layer 0. At this point I'm just happy the keyboard works again. Close enough, guess I'll just deal with it.

HaynesDC5, 2023-09-11:
> Most Keychron keyboard have a Mac/Windows switch, which changes the default layer from 0 to 2 (or from 0 to 1 on some boards). If you don't have any use for that switch, you may flip it to the Mac position (so that the default layer would be 0), and then you will be able to use all 4 layers in VIA.

Yes, so it does give me access to layers 0/1 and 2/3 depending on which system is selected but these flipped for some reason after the flash. Again, I'm just happy it works at this point.

keyboard-magpie, 2023-12-13:
Closed as completed

closed 2023-12-13 by keyboard-magpie