qmk/qmk_toolbox #412: [macOS] Need to switch from altool to notarytool

nooges, 2023-06-14:
`altool` has been deprecated in favor of `notarytool` for notarization. Need to find a replacement GitHub action that uses `notarytool` before November 1, 2023.

Possible replacements that use `notarytool`:
- https://github.com/lando/code-sign-action
- https://github.com/cocoalibs/xcode-notarization-action

cakeface, 2024-02-14:
I'm having a hard time running qmk_toolbox on OSX 14.3 due to code signing. Guessing this^^ is the issue? I'm going to try building from scratch to see if that solves the issue.

fauxpark, 2024-02-15:
The Toolbox is not codesigned. It was at one point but this was never released as the GitHub Actions that made it work were deleted, so it was removed.

tzarc, 2026-08-19:
Fixed by https://github.com/qmk/qmk_toolbox/pull/511

closed 2026-08-19 by tzarc