qmk/qmk_toolbox #416: Not flashing

amowry, 2023-07-11:
I'm not sure if this is a bug or something wrong with the configuration, but one of my customers is having an issue where a Caterina device isn't flashing. I've added a screenshot of the output below. One difference I see from my machine (where I am able to flash successfully is that there is no reference to the COM port at the very end of the "Caterina device connected" line.

[image]

fauxpark, 2023-07-11:
This is a known issue that's difficult to solve. The Toolbox detects bootloaders by watching for USB device events, and if the bootloader type in question is marked as a serial device, it then goes looking for the COM port. However these are separate entities (the USB device as a whole, and the actual serial interface) which may not necessarily appear simultaneously, and the COM port may not appear at all in some circumstances. So for now, you basically just have to keep trying until the Toolbox grabs the COM port.

amowry, 2023-07-11:
Okay, thanks! Just so I understand, it is because the device initially enumerated as both a composite device and an audio device, or it is unrelated to that? I don't recall seeing this issue before.

fauxpark, 2023-07-11:
It's unrelated. Essentially the problem is that the Windows code for the Toolbox, on detecting a caterina bootloader, makes only a single attempt to find a COM port associated with the USB device, in the constructor of the `CaterinaDevice` class. Which isn't ideal in a couple of ways.

I just haven't had the time to rework this code so that it at least tries a few more times before giving up, but I will get to it soon.

amowry, 2023-07-11:
Okay, thanks so much for your help!

amowry, 2023-07-24:
Is this issue something that just began occurring recently? I'm suddenly hearing of more users having this problem, so it almost seems like it just started happening.

amowry, 2023-07-25:
My customer is telling me that he's tried repeatedly and the upload is never successful-- is there a chance that this a different issue? Thanks!
[image]

fauxpark, 2023-07-25:
No, this is the same problem, and it has been a problem for ages.

amowry, 2023-07-25:
Okay, thanks-- do you know why it occurs on some machines and not others?

fauxpark, 2023-07-25:
Not really, it's essentially a race condition. It probably depends on the speed of your computer to some extent - if it's slower that might give the OS more time for the serial port to appear before the Toolbox gets the USB event and instantly looks for it. So it's difficult to debug, and I can't repro on my own machine.

amowry, 2023-07-26:
Okay, thanks. I can't reproduce it on my machine either.

closed 2023-10-07 by fauxpark