qmk/qmk_toolbox #417: Clear EEPROM is grayed out after DFU is connected

lesca, 2023-08-02:
Hello, 
I don't understand why the `Clear EEPROM` option is grayed out even if DFU is connected and `Flash` button is available. see figure below. 

My purpose is to reset the keymaps. This cannot be done directly use `via` firmware, but the officail firmware provides the `Fn+J+Z` key combianation to reset the keymaps. However, I guess this should also be done by clearing the EEPROM, which is not available here. 

Keyboard: Keychron Q10 
qmk_toolbox version I've tried: stable 0.2.2 , and latest beta - both don't work. 

_Figure: latest beta - Clear EEPROM is grayed out_
[image]

tzarc, 2023-08-02:
Clearing EEPROM is not supported on Toolbox with STM32 devices.

lesca, 2023-08-02:
Hello @tzarc , thanks for quick reply. I don't know much about the STM32. You make me know more. 

I want to find a way to conveniently reset the configuration if possible.

I noticed it will reset the keymap configuration to defaults if I use a firmware from different provider. 
For example, I flash the `via` offical firmware, load my saved layout; and later, I flash my custom firmware, the keymap will be reset.  However, it won't reset if I re-compile my custom firmware with modifications and flash it again. 

However, it seems too complex. It would be appreciated if you have any other ideas.

lesca, 2023-08-02:
Hello,
I find a way to make it.

QMK provides a keycode `QK_CLEAR_EEPROM` for clearing the EEPROM:

souce: [QK_CLEAR_EEPROM] (https://github.com/the-via/app/blob/bae0a0f302af525842e209f5af490959b955bb73/src/utils/key-to-byte/v12.ts#L444C3-L444C18)

the VIA web app also supports this. By assigning this keycode to a key, it successfully reset the configuration. 

Cheers!