qmk/qmk_toolbox #424: Flash button is disabled

rickyManalo, 2023-09-14:
## Describe the Bug

After following the instructions [here](https://www.keychron.com/blogs/archived/how-to-factory-reset-or-flash-your-qmk-via-enabled-keychron-c2-pro-keyboard), the Flash button seems to still be disabled. QMK already detected the keyboard as well. It already shows 

> HID console connected: Keychron Keychron C2 Pro (3434:0520:0101)

## System Information

 - Operating system: Windows 10
 - QMK Toolbox version: 0.2.2

fauxpark, 2023-09-14:
The latest Toolbox release currently does not include WB32 bootloader support. You can try the beta or use Keychron's version instead.

rickyManalo, 2023-09-14:
Can you point me to those apps?

fauxpark, 2023-09-14:
https://github.com/qmk/qmk_toolbox/releases

rickyManalo, 2023-09-14:
I tried the beta version, and it doesn't seem to detect the keyboard now

fauxpark, 2023-09-14:
You need to put the board into bootloader mode. The `HID console connected` line is unrelated, it just means the Console feature is enabled in the firmware.

rickyManalo, 2023-09-14:
Hmm, how do I do that? The guide from Keychron only said the hit fn + J + Z to go to DFU mode

fauxpark, 2023-09-14:
Try holding down the top left key while plugging it in.

rickyManalo, 2023-09-14:
You mean the esc key? The guide said that I need to hold the reset button. Can I not do that and just hold the esc key?

fauxpark, 2023-09-14:
Yes, this is the Bootmagic feature, which checks a specific matrix location (usually `0,0` - hence why I said the top left key and not Escape) on startup to determine whether to jump to the bootloader.

rickyManalo, 2023-09-15:
It seem to do something, but QMK didn't show that the board went to bootloader mode. The keyboard just didn't turn on the lights

fauxpark, 2023-09-17:
The Toolbox may not work for this bootloader anyway - it uses an older version of dfu-util which has an issue with the L4 series, IIRC. I'm trying to get a new Toolbox released soon, but you may have to just use the CLI method of flashing for now... as long as your dfu-util version is fairly recent.

rickyManalo, 2023-09-20:
I see. Well, I'm really not in a hurry. I just wanted to change my lighting
and found out that my board is having some issue with VIA. Do let me know
once you've that new build so I can test it out.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>