qmk/qmk_toolbox #430: "Unhandled exception has occurred in your application."

AwakeJJ, 2023-11-07:
Hello I've been having this issue with this popup 
[image] 
 
If you're interested in the details i put it in a txt file
[Microsoft NET.txt](https://github.com/qmk/qmk_toolbox/files/13284561/Microsoft.NET.txt)

Im getting the same thing when i try to look at my HID console
[image]

and here is that ones txt file
[QMK tool.txt](https://github.com/qmk/qmk_toolbox/files/13284586/QMK.tool.txt)

running it as admin and reinstalling drivers did nothing for me and now I'm just trying to fix my stuff

OS: Window 11
KB: KBD67 REV2

fauxpark, 2023-11-08:
Please paste the details here.

AwakeJJ, 2023-11-08:
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

 Exception Text 
System.Runtime.InteropServices.COMException (0x80040154) Class not registered

   at System.Management.ThreadDispatch.Start()
   at System.Management.ManagementScope.Initialize()
   at System.Management.ManagementEventWatcher.Initialize()
   at System.Management.ManagementEventWatcher.Start()
   at QMK_Toolbox.HidConsole.HidConsoleListener.Start()
   at QMK_Toolbox.HidConsole.HidConsoleWindow.HidConsoleWindow_Load(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, WM msg, IntPtr wparam, IntPtr lparam)

 Loaded Assemblies 
System.Private.CoreLib
    Assembly Version 6.0.0.0
    Win32 Version na
    CodeBase 
----------------------------------------
qmk_toolbox
    Assembly Version 0.2.2.0
    Win32 Version na
    CodeBase 
----------------------------------------
System.Runtime
    Assembly Version 6.0.0.0
    Win32 Version na
    CodeBase 
----------------------------------------
System.Threading
    Assembly Version 6.0.0.0
    Win32 Version na
    CodeBase 
----------------------------------------
System.Collections
    Assembly Version 6.0.0.0
    Win32 Version na
    CodeBase 
----------------------------------------
System.Windows.Forms
    Assembly Version 6.0.2.0
    Win32 Version na
    CodeBase 
----------------------------------------
System.ComponentModel.Primitives
    Assembly Version 6.0.0.0
    Win32 Version na
    CodeBase 
----------------------------------------
System.Windows.Forms.Primitives
    Assembly Version 6.0.2.0
    Win32 Version na
    CodeBase 
----------------------------------------
System.Runtime.InteropServices
    Assembly Version 6.0.0.0
    Win32 Version na
    CodeBase 
----------------------------------------
[... 217 lines cut ...]
    Win32 Version na
    CodeBase 
----------------------------------------
System.IO.Compression
    Assembly Version 6.0.0.0
    Win32 Version na
    CodeBase 
----------------------------------------
HidLibrary
    Assembly Version 3.2.49.0
    Win32 Version na
    CodeBase fileCProgram%20FilesQMK%20ToolboxSystem.Private.CoreLib.dll
----------------------------------------
netstandard
    Assembly Version 2.1.0.0
    Win32 Version na
    CodeBase 
----------------------------------------
System.Text.Encoding.CodePages
    Assembly Version 6.0.0.0
    Win32 Version na
    CodeBase 
----------------------------------------

 JIT Debugging

fauxpark, 2023-11-08:
This is an issue with your Windows install, not the Toolbox.

closed 2023-11-08 by fauxpark