qmk/qmk_toolbox #441: [HELP] Need a help my keychorn

hryer, 2023-12-17:
Hello, I apologize if this isn't the right place to raise an issue.

I'm having trouble with my Keychron Q3 QMK Custom Mechanical Keyboard's lighting. It's currently not working. When I checked VIA, I encountered several errors, as shown in the pictures below. I'm seeking guidance on how to resolve this issue. Is a factory reset of my keyboard necessary? While the keys themselves function perfectly, the lighting isn't turning on at all.

[image]
[image]

keyboard-magpie, 2023-12-17:
This should be taken up with the via team on their discord server: https://discord.gg/4fMw7YkDfY

closed 2023-12-17 by keyboard-magpie

hryer, 2023-12-18:
okay, thank you for the information