qmk/qmk_toolbox #443: Think 6.5 v2 hotswap firmware: KC_BSPC and KC_RSFT keys don't work

americanhanko, 2024-01-06:
I recently replaced my Think 6.5 v2 solder PCB with new hotswap PCB after my solder PCB stopped working suddenly. You can read my post in the Graystudio discord [here](https://discord.com/channels/503768726332506133/701193561994952785/1189693948873158716). After replacing the PCB, I moved onto installing QMK using my previous config.

In my QMK config JSON, I updated the `keyboard` value accordingly:

```diff
-    "keyboard": "gray_studio/think65/solder",
+    "keyboard": "gray_studio/think65/hotswap",
```

After flashing, I found the backspace (KC_BSPC) and left shift (KC_RSFT) didn't work at all. Keycodes showed no activity, and neither did QMK configurator in the test panel.

I thought that maybe something was different with the configuration JSON, so I rebuilt the same configuration from scratch using QMK configurator. Same results.

Finally, I selected the default firmware for the hotswap PCB, and flashed again, which resulted in the same behavior.

Only after switching back to solder PCB were the backspace and left shift functionalities restored.

## System Information

```
❯ sw_vers
ProductName:		macOS
ProductVersion:		14.2.1
BuildVersion:		23C71
```

- QMK Toolbox version: 0.2.2

americanhanko, 2024-01-06:
Incorrectly opened this in QMK toolbox. Moving to QMK firmware repo.

closed 2024-01-06 by americanhanko

keyboard-magpie, 2024-01-06:
This is not a toolbox issue. Suggest you reach out for support on the QMK discord server.