qmk/qmk_toolbox #444: Delayed debug output

jkavalec, 2024-01-19:
When I use dprintf statement to output some debug information, this is always delayed.

Example:
- have debug statement right before tap_code16()
- I press the action
- In QMK toolbox I immediately see that some keys were pressed (via those matrix outputs, that I would like to disable and see only my debug statements, but that is another matter)
- Keyboard issues the command and I see it in my editor immediately
- since the tap_code16 was already executed, I would expect my debug statement being printed immediately, this is sadly not so
- I then bash keyboard randomly and sometimes sooner sometimes later , 2-10 presses I'd say, the print statement finally shows in the Toolbox

What could be the problem? I had this happening on both Mac and Windows, and it makes debugging pretty hard

fauxpark, 2024-01-19:
Make sure your print statements end in a newline character. The toolbox will not output anything until it receives a complete line.

jkavalec, 2024-01-20:
Thank you for good advice.

This did help, and it is now usable, yet the output is always delayed by 1 press, so when action takes place and print statement is issued, it always takes one additional press to get the output.

fauxpark, 2024-01-22:
Impossible to tell what's wrong without seeing your code.