qmk/qmk_toolbox #453: Avast One Alerts

RickSeiden, 2024-02-26:
Hello,

I downloaded v0.3.0 qmk_toolbox.exe from right here on Github and when I ran it, Avast One went crazy telling me all kinds of files are infected with Win64:EVO-gen [Trj].  I **know** that this is not intentional, and I **think** this is probably a bunch of false positives, but I thought I should report it anyway.  I'm going to try downloading the installer.

[image]

fauxpark, 2024-02-26:
Yes, it is a false positive. The flashing tools bundled with the Windows Toolbox are taken directly from the MSYS2 MINGW64 packages, unmodified (you can check the hashes of the .exe if you want): https://packages.msys2.org/package/mingw-w64-x86_64-dfu-programmer?repo=mingw64

So unless someone has managed to sneak some malware into dfu-programmer itself, or replace the binary in the MSYS2 repo, it should be safe.

RickSeiden, 2024-02-27:
I ran the installer instead and had no issues with that.  I know that QMK is a very well respected project, so I don't believe for even a nanosecond that anyone here would intentionally infect their code with a virus.  Avast One, however, not so much.  Because it blocked programs that had a false positive, I couldn't use it if I wanted to.