qmk/qmk_toolbox #457: Library not loaded: @rpath/libavrdude.1.dylib

FlorianKrug, 2024-02-29:
## Describe the Bug

I want to flash my firmware to the atmega32u4, but I get the following error.

`Attempting to flash, please don't remove device
> avrdude -p atmega32u4 -c avr109 -U flash:w:/Users/floriankrug/qmk_firmware/gen1_pad_default.hex:i -P /dev/cu.usbmodem1201 -C avrdude.conf
> dyld[11970]: Library not loaded: @rpath/libavrdude.1.dylib
>   Referenced from: <EA78E0C3-114E-3D12-A05D-C6BC5EE883BF> /Applications/QMK Toolbox.app/Contents/Resources/avrdude
>   Reason: tried: '/Applications/QMK Toolbox.app/Contents/Resources/../lib/libavrdude.1.dylib' (no such file), '/Applications/QMK Toolbox.app/Contents/Resources/../lib/libavrdude.1.dylib' (no such file), '/usr/local/lib/libavrdude.1.dylib' (no such file), '/usr/lib/libavrdude.1.dylib' (no such file, not in dyld cache)
Flash complete`

## System Information

 - Operating system: MacOs 14.3.1
 - QMK Toolbox version: 0.3.0

closed 2024-02-29 by FlorianKrug

arhimmel, 2024-03-06:
i was wondering when this fix will be released?

fauxpark, 2024-03-09:
@arhimmel I have just released 0.3.1 which includes the fix from the above linked PR. If you are using the Homebrew Cask, you'll need to wait for it to be updated :)