qmk/qmk_toolbox #462: Error when flashing kb

Leandroqo, 2024-03-13:
## Describe the Bug
I'm encountering an error when attempting to flash my crkbd. Any ideas on how to resolve it?

* QMK Toolbox 0.3.1 (https://qmk.fm/toolbox)
* Supported bootloaders:
*  - ARM DFU (APM32, Kiibohd, STM32, STM32duino) and RISC-V DFU (GD32V) via dfu-util (http://dfu-util.sourceforge.net/)
*  - Atmel/LUFA/QMK DFU via dfu-programmer (http://dfu-programmer.github.io/)
*  - Atmel SAM-BA (Massdrop) via Massdrop Loader (https://github.com/massdrop/mdloader)
*  - BootloadHID (Atmel, PS2AVRGB) via bootloadHID (https://www.obdev.at/products/vusb/bootloadhid.html)
*  - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
*  - HalfKay (Teensy, Ergodox EZ) via Teensy Loader (https://pjrc.com/teensy/loader_cli.html)
*  - LUFA/QMK HID via hid_bootloader_cli (https://github.com/abcminiuser/lufa)
*  - WB32 DFU via wb32-dfu-updater_cli (https://github.com/WestberryTech/wb32-dfu-updater)
*  - LUFA Mass Storage
* Supported ISP flashers:
*  - AVRISP (Arduino ISP)
*  - USBasp (AVR ISP)
*  - USBTiny (AVR Pocket)
* Auto-Flash enabled
Caterina device connected: Arduino LLC Arduino Leonardo (2341:0036:0001) [/dev/cu.usbmodem1101]
Attempting to flash, please don't remove device
> avrdude -p atmega32u4 -c avr109 -U flash:w:/Users/leandro.queiroz/qmk_firmware/crkbd_rev1_leandroqo.hex:i -P /dev/cu.usbmodem1101 -C avrdude.conf
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:399]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:400]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:428]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:429]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:461]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:462]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:495]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:496]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:522]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:523]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:563]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:564]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:583]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:584]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:602]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:603]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:648]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:649]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:732]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:734]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:743]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:745]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:755]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:757]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:768]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:770]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:791]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:793]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:815]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:817]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1184]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1185]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1195]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1197]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1218]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1219]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1236]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1237]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1252]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1253]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1263]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1264]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1274]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1275]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1285]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1286]
> avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1302]
> avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/Applications/QMK Toolbox.app/Contents/Resources/avrdude.conf:1306]
[... 122 lines cut ...]
Ψ Repo version: 0.12.47
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-11-06 23:07:44 -0300 (a66312332c) -- crkbd leandroqo based on gotham
Ψ - Latest upstream/master: 2024-03-13 08:24:21 +0100 (aa73362674) -- [Keyboard] add glyphkbd_v2 (#23131)
Ψ - Latest upstream/develop: None
[... cut ...]
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found

right7ctrl, 2024-03-14:
I just faced the same issue on my M1. Using version 0.2.2 its worked for me

Leandroqo, 2024-03-14:
> I just faced the same issue on my M1. Using version 0.2.2 its worked for me

It's worked. Thanks 🤘

closed 2024-03-14 by Leandroqo

izaakm, 2024-03-19:
I'm also having this issue w/ v0.3.1.

I propose re-opening this because downgrading it does not resolve the issue.